riscv-non-isa/riscv-external-debug-security #10: Backward compatibility with RISC-V Debug Specification

joxie, 2024-03-26:
https://github.com/joxie/riscv-debug-security/issues/10

joxie, 2024-03-26:
I think we have updated spec to fixed this issue, closing

AoteJin, 2024-04-11:
The debugger has to go through this sequence:
>- Not Compatible with RISC-V Debug Spec: Change dscr.prv to M-mode and then read mstatus by abstract command/program buffer

The reason to choose the non-compatible way is that even when mdbgen[i]=1, the following behaviors are still not allowed:
- bypassing pmp by relaxedprv
- ndmreset

That is why compatible way seems not the best practice and the debugger can actually discover the extension first and then proceed with the knowledge whether it is compatible with debug spec.

gokhankaplayan, 2024-04-11:
Thank you for the clarification, @AoteJin.

closed 2024-04-11 by gokhankaplayan