riscv-non-isa/riscv-external-debug-security #12: Per priviledge-mode debug control

joxie, 2024-03-26:
Open this ticket to track potential need for per privilege control for external debug / trace

joxie, 2024-03-27:
@AoteJin can you update your discussion with Penlai guys

AoteJin, 2024-03-27:
Penglai team claim that their architecture does not requirement per privilege-mode debug control. They could accommodate current proposal by:
- Enhance the secure monitor to context switch strictly on submachine boundary (which aligns with their roadmap as well)
- Add supervisor RT layer for debug of u-mode only partition

gagachang, 2024-03-27:
> * Add supervisor RT layer for debug of u-mode only partition

Hi, What if CPU only implements M+U without S-mode?

gagachang, 2024-03-27:
Hello @AoteJin
Would you like to explain what `per privilege-mode debug control` means ?
I might miss the discussion in previous TG meeting.

AoteJin, 2024-03-27:
> Hello @AoteJin Would you like to explain what `per privilege-mode debug control` means ? I might miss the discussion in previous TG meeting.

I would love to. The external debug security proposal has two level control: machine (from RoT) and submachine (by M-mode CSR). 

>What if CPU only implements M+U without S-mode?

In an implementation without S-mode, these two controls are enough to separately manage debuggability of M and U. 

This issue to discuss the case that an implementation has privilege levels more than M/U (e.g. M/S/U) and machine mode launches directly to U-mode. This might be a problem because submachine mode control does not differentiate S/U and results in privilege escalation in terms external debug.

gagachang, 2024-03-27:
> > What if CPU only implements M+U without S-mode?
> 
> In an implementation without S-mode, these two controls are enough to separately manage debuggability of M and U.

Yes, U-mode external debug can be managed by `sdbgen`, in M+U system. Thanks!

> This issue is to discuss the case that an implementation has privilege levels more than M/U (e.g. M/S/U) and machine mode launches directly to U-mode. This might be a problem because submachine mode control does not differentiate S/U and results in privilege escalation in terms external debug.

What the privilege escalation is here?
Do you mean U-mode could read S-mode CSR or memory (e.g., satp, stvec) ?
Because maximum debug privilege is S-mode when `mdbgen=0` and `sdbgen=1`

AoteJin, 2024-03-27:
> What the privilege escalation is here?
Do you mean debugger could read S-mode CSR or memory (e.g., satp, stvec) ?
Because maximum debug privilege is S-mode when mdbgen=0 and sdbgen=1

Yes, this is exactly the concern.

gagachang, 2024-03-27:
Another problem is `dcsr.PRV` ?
U-mode debugger can write  `dcsr.PRV` as S-mode (Because maximum debug privilege is S-mode) and dret to S-mode, am I correct?

gagachang, 2024-03-27:
Hmm.. The hart knows the previous privilege mode (i.e., U-mode) when the hart enters Debug Mode.

In external debug spec section 3.4.1 says:
> It will generate a security fault error (cmderr 6) if the external debugger attempts to
configure prv and v with a privilege higher than the maximum debug privilege level.

Maybe it can changed to:
It will generate a security fault error (cmderr 6) if the external debugger attempts to
configure prv and v with a privilege higher than the "previous" privilege level.

AoteJin, 2024-03-27:
> I mean, when external debugger executing abstract commands to read/write CSR, memory, hart can check both maximum debug access privilege and "previous" privilege level.
For example, dcsr.PRV can not be configured to S-mode if previous privilege level is U-mode.

This is not directly related to this ticket. If you have concern on it, would better to discuss it on a separate issue.
The reason not checking against previous privilege is that the subM debugger might stop on U-mode and it will need  higher privilege "view" to debug the U-mode, e.g. VA->PA mapping etc. We do not want to break this use case.

joxie, 2024-04-18:
Closing. Based on the information TG has as of today, we do not see a need to have fine grain control.

closed 2024-08-05 by AoteJin