riscv-non-isa/riscv-external-debug-security #8: Performance Counter enable for M-mode

joxie, 2024-03-26:
Copied from https://github.com/joxie/riscv-debug-security/issues/18

@gokhankaplayan 

Performance counters has configuration registers (mcyclecfg, minstretcfg) to isolate between privilege levels. HW RoT should be able control global enable for performance counters. For example, medbgen is a global enable for debug (+trace) here: https://github.com/riscv/riscv-smmtt/blob/main/chapter8.adoc

Similar mechanism seems missing for performance counters. Should we extend the scope for performance counters?

joxie, 2024-03-26:
@gokhankaplayan can you please update this issue with threat model?

joxie, 2024-04-09:
Ping @gokhankaplayan

gokhankaplayan, 2024-04-09:
When M-mode debug is disabled, external debugger is not able to access M-mode perf counters. I think this part is already covered by the current specification. It was not my intention for this issue.
> This is an interesting topic, in general if external debugger can access performance counters when M-mode debug is disabled then there's a problem that we must fix. 
> We need to explicitly mention machine mode MMIO registers (mtime and mtimecmp) are not accessible to external debugger when M mode external debug is disabled

The threat model: Due to privilege escalation attacks, global enable mechanism of trace&debug is controlled by externally (not by M-mode CSR). I was wondering that same security measure should be taken for performance counters. But, I could not find relevant attack scenario for performance counters. If there is no objection, I am proposing to close this issue.

closed 2024-04-09 by gokhankaplayan