riscv-non-isa/riscv-external-debug-security #9: External Trigger of Core Trigger Module

joxie, 2024-03-26:
https://github.com/joxie/riscv-debug-security/issues/12

AoteJin, 2024-03-27:
Current proposal has the subsection `3.3.2. External triggers` to elaborate on external trigger.
@gokhankaplayan  could you help review whether it is sufficiently address the issues related external trigger.

AoteJin, 2024-04-09:
ping @gokhankaplayan

gokhankaplayan, 2024-04-09:
> The external trigger outputs follow the same limitations as other triggers, ensuring they do not fire or match when the privilege level of the hart exceeds the ones specified 

There is no way to ensure external triggers do not fire according to privilege level of the hart. External trigger can be from another core, external debugger, interrupt.

AoteJin, 2024-04-10:
> There is no way to ensure external triggers do not fire according to privilege level of the hart. External trigger can be from another core, external debugger, interrupt.

I think what you refers to is external trigger input. The external trigger output is just the same as other trigger which is configured via action field (action =8/9).
[image]

Regarding the trigger input, I describe it in the follow paragraph:

_"The sources of external trigger input (such as machine mode performance counter overflow, interrupts, etc.) require protection to prevent information leakage. The external trigger inputs supported are platform-specific. Therefore, the platform is responsible for enforcing limitations on input sources. As a result, tmexttrigger.intctl and tmexttrigger.select should be restricted to legal values based on mdbgen and sdedbgalw. Their definitions are provided in the Table 8 below."_

Basically, since the set of trigger input sources are platform-specific, the paragraph requires platform to secure it according to the property of the input sources. What do you think of it?

gokhankaplayan, 2024-04-10:
I was referring to the external trigger input (not its action). Restricting the legal values of tmexttrigger.select makes sense to me depending on external trigger type. I am fine with the current specification. 
Please note that I would recommend to force select to 0 when mdbgen=0 for some external trigger types, such as trigger coming from another core. If implementation prefers to check sdedbgalw for these types of triggers, it might fire or not fire depending on when trigger fires. Another option is allow the external triggers to be pending until allowable debug privilege level is entered. I know this option is not preferred right now.

AoteJin, 2024-04-12:
>Please note that I would recommend to force select to 0 when mdbgen=0 for some external trigger types, such as trigger coming from another core. 

I agree with statement, but the difficulty here is that such external trigger inputs vary among platforms and not standardized. There seems no good way to specify the rules in a consistently. 

>Another option is allow the external triggers to be pending until allowable debug privilege level is entered. I know this option is not preferred right now.

This option introduces risk of info-leak, e.g. a M-mode perf counter overflow trigger pends until enter submachine mode and fires.

AoteJin, 2024-07-02:
The latest PR #47  add the description to elaborate on that the external trigger inputs are managed by the sources from where the input signal are generated. For example, if the signal is from another hart,  whether the signal is asserted depends on whether debug is allowed for the hart.
@gokhankaplayan  could you help review the updates?

AoteJin, 2024-07-30:
Close as the PR merged to main branch

closed 2024-07-30 by AoteJin