riscv-non-isa/riscv-external-debug-security #16: Security fault reported as cmderr

vicky-goode-img, 2024-04-05:
Based on spec version _v0.0.0, 2024-03-26_ I have the following questions and note about security faults, as they can happen independent of abstract command (for example during halt on reset or hartreset) but they're logged in abstractcs.cmderr:

1. how will the external debugger know which hart the security fault applies to? 
   a) It could happen that one hart gets a security fault while another is executing an abstract command. In that case, what happens with regard to error priorities if the abstract command actually gets a non-security fault error?   
2. The cmderr is no longer restricted to only being set if an abstract command fails, which is what it’s defined as in the original debug spec.

If it’s acceptable not to know which hart the security fault applies, I think these points should be explicitly stated in the spec so that it’s clear that this has been considered.

AoteJin, 2024-04-08:
Yes, your points make sense. Raising error in abstractcs.cmderr for halt/reset/hartreset/keepalive/ndmreset operation seems not a good idea. 
Probably we could propose to add extra two bits in dmstatus (at 0x11), anysecfault, allsecfault, to indicate the error caused by aforementioned operations?

vicky-goode-img, 2024-04-09:
I think that could work. For that case I have the following questions:

1. So the DM would be expected to track a per hart bit of secfault as well as the new secfault option in cmderr and sberror?
2. If a hart got a secfault during execution of an abstract command, would that set cmderr plus the hart specific secfault?
3. How would the hart specific secfault bits be cleared? Only by resetting the DM?
4. For a secfault caused by the debugger attempting ndmreset, which isn't hart-specific, what would that set? All existent harts' secfault bits? Or some separate general secfault bit? Or something else?

joxie, 2024-04-09:
Discussed in this week's TG meeting - we will add extra bits to indicate the error caused by halt/reset/hartreset..etc

For ndmreset we probably don't need a dedicate error, debugger can always read the register and find out that the feature is not implemented.

@AoteJin  please update the proposal

closed 2024-04-18 by AoteJin