riscv-non-isa/riscv-external-debug-security #20: Add control nob to ndmreset

joxie, 2024-04-09:
Open this issue to track potential need to add control nob for ndmreset, the current proposal defeatures ndmreset.

gokhankaplayan, 2024-04-10:
ndmreset is required for the original RISC-V Debug Specification and external debuggers (openocd and etc.) uses ndmreset.
My proposal is to add external control signal which is controlled by RoT (similar to mdbgen control signal). What is effected by ndmreset is implementation specific in the original RISC-V Debug Spec. SoCs can exclude certain blocks such as RoT, Power Controller.

joxie, 2024-04-22:
@gokhankaplayan can you clarify the usage of this nob? I am actually not very sure why RoT would ever want to enable ndmreset with ext. debug. ext. implemented, one thing will likely happen is debugger will likely lost access if debug is reset to disabled.

gokhankaplayan, 2024-04-22:
In the early lifecycle of chip, such as debugging boot process, It is desirable to debug from very first instruction with **clean stytem state**. Using ndmreset provides a way to start with clean state for system components (peripherals and memory) to debug the boot process. If ndmreset causes security issues, it can be disabled by nmdreset nob as part of the lifecycle management by fusing.

AoteJin, 2024-08-27:
A system level control knob is introduced in PR #56