riscv-non-isa/riscv-external-debug-security #23: mdgben as global know instead of per-hart

gokhankaplayan, 2024-04-10:
An input port, named mdbgen[i], is introduced for each hart i in Chapter 2 (ISA-part). I propose to have global input port mdben instead of per-hart mdbgen. 
There is no need for per-hart control in homogeneous computing.

vicky-goode-img, 2024-04-10:
If this proposal is rejected and mdbgen remains per hart, then in the Debug Module (non_ISA) chapter the use of mdbgen needs to be clarified, e.g 

- for hartreset and halt it should use mdbgen[i] instead of mdbgen
- for ndmreset it should probably say "Issuing ndmreset operation results in a security fault error **_when any bit of_** mdbgen is 0"

And possibly other cases in that chapter

AoteJin, 2024-04-11:
I think we should take heterogeneous scenarios into consideration as well. A certain SoC might have many varied profile of RISC-V hart serving different purpose. And each of them can be of different security level, thus have various debug policy for M-mode. Do you think it makes sense?

gokhankaplayan, 2024-04-11:
Since the RISC-V Spec is generic, It makes sense to keep per-hart mdbgen[i] input in the Spec. Homogenous scenarios can implement global mdbgen input as an implementation choice. 
As @vicky-goode-img mentioned we should be consistent to use mdben[i] in the document.

closed 2024-04-18 by AoteJin