riscv-non-isa/riscv-external-debug-security #27: ecall/xret behaviour in debug mode

gokhankaplayan, 2024-04-30:
The RISC-V Debug Spec has the following statement about ecall/xret behaviour in debug mode: 

> Almost all instructions that change the privilege level have unspecified behaviour. This includes ecall, mret, sret, and uret.

The question is whether Zedsec ISA extension should restrict their behaviour. Should executing ecall/xret in debug mode result in security fault error (abstractcs.cmderr=6)? 

My answer is NO. Zedsec ISA extension does not have to restrict behaviour (unspecified behaviour). How to change the privilege level is already clarified in the RISC-V Debug Spec and Zedsec ISA extension:

> To change the privilege level, the debugger can write prv and v in dcsr.

I think it is legal if some implementation treat ecall/xret in debug mode as nop or exception (abstractcs.cmderr=3).

pdonahue-ventana, 2024-04-30:
Note that UNSPECIFIED doesn't guarantee the security properties that you want.  See https://lists.riscv.org/g/security/message/976

Maybe the secure debug spec needs to say something about this or maybe there will be an overall requirement added to the definition of UNSPECIFIED.  But you should be aware of the issue and make sure that it is resolved to your satisfaction.

gokhankaplayan, 2024-05-01:
I agree that UNSPECIFIED can lead to any implementation including changing the effective privilege level. So, Zedsec ISA extension should explicitly specify that executing ecall/xret in debug mode leads to exception (abstractcs.cmderr=3) or treated as NOP. IMO it is implementation specific which option (exception or NOP) is implemented.

pdonahue-ventana, 2024-05-01:
I agree.  Your proposal allows some implementation flexibility while maintaining security against malicious debuggers.

joxie, 2024-05-07:
Agree that we need to specify the behavior in spec to avoid ambiguity and potential side effects.

AoteJin, 2024-05-10:
Close the issue with PR #32

closed 2024-05-10 by AoteJin