riscv-non-isa/riscv-external-debug-security #29: Naming of Zedsec

pdonahue-ventana, 2024-04-30:
The unprivileged spec says:
> The first letter following the "Z" conventionally indicates the most closely related alphabetical extension category, IMAFDQCVH.

Because of the second letter, Zedsec would be an extension related to E, the base architecture that contains only 16 registers.  That's not correct.  Debug-related extensions start with Sd (such as Sdext and Sdtrig).  To comply with the naming convention, I suggest renaming Zedsec to Sdsec.

gokhankaplayan, 2024-05-01:
Thank you for the detailed explanation. Sdsec seems more appropriate according the RISC-V ISA convention. What do you think @joxie ?

joxie, 2024-05-02:
Makes sense, https://github.com/riscv-non-isa/riscv-external-debug-security/pull/30 created to fix the naming

closed 2024-05-02 by joxie