riscv-non-isa/riscv-external-debug-security #35: aamvirtual restriction is not required

gokhankaplayan, 2024-05-23:
The following restriction about aamvirtual does not make sense to me for two reasons:
> The field aamvirtual in command (at 0x17) determines whether physical or virtual address translation is employed. However, when mdbgen[i] is 0, the extension mandates that aamvirtual is hardwire to 1 and memory access addresses are processed as if initiated by the hart in debug access privilege.

- Assuming mdbgen[i]= 0 and sdedbgalw=1, The external debugger is able to access satp CSR. So, it can control (disable) the virtual address translation. 
- This restriction is not necessary since we want to protect M-mode from a Supervisor Domain. It is protected by PMP (or its equivalents like MTT) not by virtual address translation. If we had a requirement to protect Supervisor Mode from User Modes, this requirement would make sense (We need user mode debug enable if we had this requirement). 

As a result, I proposed to remove this restriction from the spec.

AoteJin, 2024-06-03:
We need to consider the corner case when mstatus.TVM is 0x1. In this case, the virtual translation cannot be disabled in S-mode. 
If we remove aamvirtual restriction, there will be a hole when mstatus.TVM is 0x1.

gokhankaplayan, 2024-06-03:
Thank you for the explanation @AoteJin. I was not aware of TVM bit setting. I am closing the ticket.

closed 2024-06-03 by gokhankaplayan