riscv-non-isa/riscv-external-debug-security #37: spec v0.5 ch 4 - end feedback

bcstrongx, 2024-05-31:
By chapter

4.5
* Typo "details" -> "detailed", and "constrains" -> "constraints"

4.7 
* I think you probably want a dmcontrol.acksecfault bit, to clear the any/allsecfault bits for selected harts.  Otherwise an S-mode debugger can't ever clear them, since it can't ever execute a successful resethaltreq, reset, or keepalive.

4.8
* Register 1 caption should probably just say "new bits in dmstatus" or the like, since it has more than just the *secured bits
* Table 8: resize column widths

AoteJin, 2024-06-03:
>I think you probably want a dmcontrol.acksecfault bit, to clear the any/allsecfault bits for selected harts. Otherwise an S-mode debugger can't ever clear them, since it can't ever execute a successful resethaltreq, reset, or keepalive.

I am wondering whether there is use case to ack the fault. Because if it is not possible to issue successful operation, it will be fine to always indicate error.  And the error just represent the result of the last operation, I don't think there is ambiguity even if it is not cleared.

AoteJin, 2024-06-04:
fixes applied in #38, closing

closed 2024-06-04 by AoteJin