riscv-non-isa/riscv-external-debug-security #48: Elaborate on trace related sideband signal

AoteJin, 2024-07-23:
- Document the desired trace behavior relative to the trace sideband signal values.
- Align the naming of the signals

bcstrongx, 2024-07-23:
In the discussion linked above, I made two suggestions:

1) Remove any reference to the hart-to-encoder interface (including new signals proposed for it) from the Sdsec spec, since it is purely an implementation detail and there is no mandated implementation.  An architecture spec should refer only to behavior based on architectural values (like mtrcen and sdetrcalw).  The interface is also being removed from the trace spec, and moved to a whitepaper.

2) In the hart-to-encoder whitepaper, call the new signal sec_inhibit rather than sec_check, since it serves to inhibit trace when set.

AoteJin, 2024-07-30:
The sdsec will follow suggestion in https://lists.riscv.org/g/tech-external-debug-security/message/124 to add descriptive content on architectural state, leaving out the interface details.

AoteJin, 2024-08-06:
Hi Beeman, I updated the description of the trace in PR #53. Could you help review it?

AoteJin, 2024-08-13:
closing with PR #53  submitted.

closed 2024-08-13 by AoteJin