riscv-non-isa/riscv-external-debug-security #49: Trigger registers accessibility in debug mode

AoteJin, 2024-07-23:
Per the current definition, the trigger registers are always accessible in debug mode to allow a debugger with S-mode privilege to set up triggers. This scheme might introduce extra hardware complexity since the trigger registers bypass privilege-based CSR access rule checks in debug mode. There are other candidates for trigger usage:

- The debugger sends a message to M-mode software to set up the trigger.
- Add a set of shadow CSRs in S-mode to allow the debugger to set triggers.

It is desired to choose a scheme that balances hardware complexity and usability.

bcstrongx, 2024-07-23:
The first option, a "DBI" (debug binary interface), seems complex.  I'm not sure how the debugger could invoke M-mode to configure the triggers.  There are no traps from debug mode today, and the debugger can't set dcsr.prv=M when mdbgen=0.

The second option wouldn't be that hard.  There would be no new CSRs, just a set of indirect CSR indices that allow S-mode to access the trigger CSRs via siselect/sireg* (minus the m bits).  And there would be an Smstateen bit that controls S-mode access to these indirect CSRs.  There would be something similar for VS-mode.  There are several extensions that have already done something similar, so as long as the definition hews closely to those it should be non-controversial.  I could help with this part.

bcstrongx, 2024-07-25:
I'll add that the second option would also benefit in-target debuggers, eliminating the need for an SBI (and the associated overhead) to use triggers.

AoteJin, 2024-07-30:
Regarding option 2, the overhead for debugger is to set debug privilege to S-mode (when halted in U-mode), and program the delegated trigger. But it isnt heavy. 
It seems delegating trigger to S-mode could be a separate extension to debug spec and the main use case for it is to enable OS to set up triggers fast and external debug security also benefits from it.  Probably it's better to have another spec for it instead of adding it in external debug spec if we opt for this approach.

AoteJin, 2024-07-31:
Yes, that would be great. Thank you for taking the initiative!

AoteJin, 2024-09-23:
close the issue as Smtdeleg/Sstcfg proposed by Beeman was approved by as a fast-track: https://lists.riscv.org/g/tech-external-debug-security/topic/risc_v_tech_chairs_arc/108528941

closed 2024-09-23 by AoteJin