riscv-non-isa/riscv-external-debug-security #50: Quick Access behavior when debug is disallowed

AoteJin, 2024-07-25:
Quick Access is a special abstract command which performs the sequence of operations: halt, execute program buffer and resume.  There are several candidates to handle it when debug is disallowed:
- Let halt operation pend like other kind of halt. 
     - The debugger needs to reset itself when the QA is issued to a debug-disallowed hart.
- QA responds with an error when mdbgen[i] is 0
    - QA is disallowed for S-mode debugger 
- QA responds an error when exceed preset timeout 
- QA responds with an error when hart runs in a debug-disallowed privilege 
     - Hart needs to differentiate hart from QA and other halt

gokhankaplayan, 2024-08-06:
Let me comment on them according to the solution list you proposed:

1. Unlike a halt request, there is no mechanism to cancel quick access. A pending option will block the debugger, and only a reset can resolve this issue. However, I'm not sure if a reset is an acceptable solution from the debugger's perspective.
2. This is a simple solution for both the external debugger and hardware, but it limits the quick access use case.
3. There is extra hardware complexity and a question about how to set the timeout value. It heavily depends on the use case.
4. Differentiating a halt request and providing an error response from the core perspective will be difficult to implement from a hart perspective.
In summary, I am in favor of the second option due to the fact that the use case of quick access is not clear in my mind for S-mode debuggers.

AoteJin, 2024-08-27:
The TG votes for the second solution. The spec is updated in PR #56

closed 2024-08-27 by AoteJin