riscv-non-isa/riscv-external-debug-security #51: dmcs2.acksecfault "new value" of hartsel and hasel clarification

vicky-goode-img, 2024-08-02:
The dmcs2.acksecfault field description (table 8) has the following statement: "Writes apply to the new value of hartsel and hasel." This clarification seems like a mistaken copy paste out of the dmcontrol register fields like haltreq and resumereq. Because the acksecfault bit is not part of dmcontrol, hartsel and hasel cannot change in the same cycle as this is written, so there is no "new value" of hartsel and hasel when acksecfault is written.

gokhankaplayan, 2024-08-02:
I guess "new value" part is added by mistake here. What do you think @AoteJin?

AoteJin, 2024-08-05:
Thanks for spotting it. It is added by mistake, will remove it.

AoteJin, 2024-08-05:
Close with update of the description in table 8.

closed 2024-08-05 by AoteJin