riscv-non-isa/riscv-external-debug-security #92: Debug Module Security Extension and Sdsec do not seems to support Machine + User CPUs

Silabs-ArjanB, 2025-04-29:
Hi,

"Table 1 External Debug Configuration and Privilege" shows that _mdbgen_ and _sdedbgalw_ can be used to configure the debug access privilege to M-mode or S-mode. There is no mechanism to set debug access privilege to User mode, which would be useful on CPUs that only provide Machine and User mode (and not Supervisor mode). On such CPUs the _sstatus_, _hstatus_, _sdcsr_, _sdpc_, etc. registers would not exist either and it would not be correct to use Supervisor privilege for debug accesses in that scenario.

Can Machine + User CPUs be supported as well with equivalent features as currently described for Machine + Supervisor CPUs?

Best regards,
Arjan

AoteJin, 2025-05-14:
Hi Arjan,

When the extension was first initiated, we did not receive much requirement from M+U use cases. M-mode undertakes non-trivial workload in such design, analog to a pseudo OS. Thus, we believe the M-mode can also fit in a self-hosted debugger to securely debug U-mode applications. 

Could you please elaborate on your use case and the requirement for external debug isolation between M/U? E.g. M-mode code is very simple and cannot accommodate a debugger, but also sensitive and cannot be exposed when debugging U-mode.

joxie, 2025-05-14:
Good question. I think M/U mode is also supported with current spec with minor modification, M-mode can enable/disable U mode debug via sdedbgalw, the max debug privilege would be U instead of S now.

Silabs-ArjanB, 2025-05-14:
> Could you please elaborate on your use case and the requirement for external debug isolation between M/U? E.g. M-mode code is very simple and cannot accommodate a debugger, but also sensitive and cannot be exposed when debugging U-mode?

My question is motivated by the desire for (debug) security even in low end (low cost, low performance) devices where adding Supervisor mode is not needed and deemded to expensive, but we still use M+U.

joxie, 2025-05-21:
Hi @Silabs-ArjanB  we discussed this topic in last night's TG meeting, there're two choices

1. M-mode can provide indirect debug support (gdb stub) to U mode, there's no need to change spec 
2. If the use case is strong enough to justify that 1 is not good enough then we will need to decide the scope of change, is it enough with external debug via sdedbgalw and sdetrcalw, or external debug + trigger delegation, the problem for triggers is that Smtdeleg required by Sdsec does not support U-mode delegation so the scope of this change maybe large.

In the end of the day the support for M/U cores isn't something that will naturally happen, it needs some justification. Many extensions do not support M/U cores either, such as Smtdeleg and also Smmtt. I will start a thread in the TG mailing list to collect feedback.

joxie, 2025-07-01:
Closing, the current decision is prioritize M/S/U support for the first ratified version in order to avoid spec rework.