riscv-non-isa/riscv-external-debug-security #94: Internal review (v0.7.1) feedback

bcstrongx, 2025-05-09:
General:
* There are several extra blank pages
* Probably want to say that, if Sdsec and Sdtrig are implemented then Smtdeleg/Sstcfg and Sdtrigpend are required.  I now see this in 3.4.2, but I would phrase it as I did here.
* I assume Sdsec depends on Smsdedbg?  And Smsdetrc, if trace is implemented?
* RISC-V specs generally refer to CSRs and their fields as `csrname`.FIELDNAME (CSR name is lower case and fixed-width font, field name is upper case).  I realize that the debug spec does not do this, but it is the exception, since it was started over 4 years ago, before some of the current standards were in place.  I'd recommend that this spec adopt the current standards.

1.0: 
* "Triggers: Triggers (with action=1) can only fire or match when external debug is allowed in the
current privilege mode." with Sdtrigpend triggers can match if the effective priv mode of the access is debug-allowed, but will pend the action rather than firing.

1.1:

-  "The privilege with which an Abstract Command or instructions in the Program Buffer
access hardware resources" -> "The privilege **level** with which an Abstract Command or instructions in the Program Buffer
access**es** hardware resources"

3.1:
* Same comment about trigger match/fire.

3.2:
* This sentence is confusing: "The availability of trace output is indicated through the hart-trace interface (HTI) [2]"  The HTI is pure implementation/uarch, and not required.  So I would remove this sentence, and instead add a non-normative comment saying that mtrcen and sdetrcalw could be signals added to the HTI.

3.3 and 3.3.2:
* Same comment about trigger match/fire.

3.4.2:
* This spec does not extend Sdtrig, so I would name this section "Trigger Access", or the like.

3.4.3:
* `dcsr` is the Debug Control and Status Register, so I would call msdcfg something more like the Machine Supervisor Debug Configuration Register.

4.3:
* The hartreset behavior seems odd.  There's no way for the debugger to know when the hart is in M-mode.  Would it not make more sense for the reset to remain pending?  That puts more burden on the hart, but what if the hart spends 90% of the time in M-mode?  The debugger might never be able to reset it.  Or maybe it could first halt the hart (which would be pended until it can be taken), then set hartreset?

AoteJin, 2025-05-13:
Thanks Beeman!

> There are several extra blank pages

It seems to be caused by RISC-V themes, maybe we should follow it?

> I assume Sdsec depends on Smsdedbg? And Smsdetrc, if trace is implemented?

Right. Do you think the spec needs explicitly mention that Smsdetrc is required when trace is implemented?

> RISC-V specs generally refer to CSRs and their fields as csrname.FIELDNAME (CSR name is lower case and fixed-width font, field name is upper case). I realize that the debug spec does not do this, [...]

Sure, will change the style after updating the normative text. 

> "Triggers: Triggers (with action=1) can only fire or match when external debug is allowed in the
current privilege mode." with Sdtrigpend triggers can match if the effective priv mode of the access is debug-allowed, but will pend the action rather than firing.

Explicitly added effective privilege level for trigger matching or firing. Do you think it is better?

> This spec does not extend Sdtrig, so I would name this section "Trigger Access", or the like.

Changed to Trigger Security and Access

> so I would call msdcfg something more like the Machine Supervisor Debug Configuration Register.

Actually, msdcfg is the CSR defined in Smmtt. The Sdsec just add a field to this register since the security control is applied on supervisor domain boundary.

> The hartreset behavior seems odd. There's no way for the debugger to know when the hart is in M-mode.

The hartreset will reset the hart when M-mode is debug-allowed regardless of the current hart privilege level. When M-mode is debug-disallowed, the hartreset results in secfault error. The debugger doesnt need to wait till the hart enters M-mode. Does it sounds straightforward to you?

Other points are simply accepted and updated in https://github.com/riscv-non-isa/riscv-external-debug-security/commit/b544205a93cdf3392cd17bcc4a7730a4ae2766a8

bcstrongx, 2025-05-13:
> It seems to be caused by RISC-V themes, maybe we should follow it?

Hmm, I guess so.  Maybe make sure you update submodules to see if there are updates that would reduce the waste?

> Right. Do you think the spec needs explicitly mention that Smsdetrc is required when trace is implemented?

I would make sure that any requirements that are not stated in other specs are stated here.

> Explicitly added effective privilege level for trigger matching or firing. Do you think it is better?

I think that works.  More details included in Sdtrigpend.

> The hartreset will reset the hart when M-mode is debug-allowed regardless of the current hart privilege level. When M-mode is debug-disallowed, the hartreset results in secfault error. The debugger doesnt need to wait till the hart enters M-mode. Does it sounds straightforward to you?

But if an S-mode debugger wants to reset the hart, it may have to repeatedly set hartreset and clear the error until it "gets lucky" and the hartreset arrives while the hart is in S-mode.  It would be nicer from a user perspective if the hartreset remains pending until it can be taken, just like haltreq.  Next best would be to recommend that the S-mode debugger first assert haltreq, which will halt the hart once it enters S-mode.  Then the debugger can assert hartreset and be assured that the reset is taken.  So if you don't want to require hartreset requests to be held pending by the hart, I'd add a non-normative comment about using the haltreq approach.

AoteJin, 2025-05-14:
> Hmm, I guess so. Maybe make sure you update submodules to see if there are updates that would reduce the waste?

Tried out with the latest themes. But seems not working ...

> I would make sure that any requirements that are not stated in other specs are stated here.

Agree. I will add the statement then.

> But if an S-mode debugger wants to reset the hart, it may have to repeatedly set hartreset and clear the error until it "gets lucky" and the hartreset arrives while the hart is in S-mode. It would [...]

An S-mode debugger cannot reset the hart even when the it is in S-mode. That is designated to avoid an S-mode debugger to replay M-mode boot code. Only an M-mode debugger can reset the hart and the hartreset can be served regardless of the current hart privilege level.

bcstrongx, 2025-05-14:
> An S-mode debugger cannot reset the hart even when the it is in S-mode. That is designated to avoid an S-mode debugger to replay M-mode boot code. Only an M-mode debugger can reset the hart and the hartreset can be served regardless of the current hart privilege level.

Oh I see, I misread it.  So no hartreset when mdbgen=0.  Okay that works.

joxie, 2025-07-01:
Closing as fixed

closed 2025-07-01 by joxie