riscv-non-isa/riscv-external-debug-security #98: Re-entry into debug mode after interrupt during single step

pmundkur, 2025-05-20:
The current text states that interrupts that trap to M-mode are not affected when `mdbgen` is 0 and `sdedbgalw` is 1.  This indicates that debug mode can be exited to M-mode during single step to service non-delegated interrupts; however, the mechanism for re-entry into debug mode to continue the single step is not clear.  It would be good to spell out the conditions under which this single step can be safely resumed, and when it cannot.

gagachang, 2025-05-21:
In your case `mdbgen=0` and `sdedbgalw=1`, I don't think the debug mode can be exited to M-mode.
Please check table 3, Maximum Allowed Resume Privilege Mode.

pmundkur, 2025-05-21:
I'm referring to the last part of  the statement in the "Interrupt during Single Stepping" section: "When mdbgen is 0 and sdedbgalw is 1, only delegated interrupts are disabled, while interrupts that trap to M-mode are not affected."    If interrupts that trap to M-mode are not affected (by disabling), does that mean that they trap to M-mode (and how does this affect Table 3)?  If they do trap to M-mode, how does the single stepping resume?

joxie, 2025-05-23:
Under this scenario that `mdbgen`=0, and `sdedbgalw`=1, a M-mode interrupt hits single-step, hardware should work as if there's no M-mode interrupt, i.e. hardware will exit debug mode, trap into M-mode, execute trap handler in M-mode, mret, execute one instruction, and finally re-enter debug mode. At this time debugger can use `resumereq` to resume execution.

pmundkur, 2025-05-23:
Thanks, that makes sense.  It would be worth spelling this out in the text, a bit like the "Single Step" section in the Debug spec but now about the `sdcsr.step` bit.   An entry could also be added to the first list ("Debug operations affected by Sdsec") in the "3.1 External Debug" section for "Single Step operation" or similar, since single step is also affected by Sdsec.

joxie, 2025-05-23:
@pmundkur please review PR #99

AoteJin, 2025-05-26:
Hi Prashanth,

> This indicates that debug mode can be exited to M-mode during single step to service non-delegated interrupts;

This description is not accurate. The debug spec says:
1.  all interrupts are masked in debug mode 
2. If step is set when a hart resumes then it will single step

That is, M-mode interrupts cannot cause the hart exiting debug mode to serve the interrupt. The trap is only taken when the hart resumes. The Sdsec follows the same behavior of debug spec. The difference lies on dcsr.stepie, while delegated interrupts may be disabled during single-stepping. 

> the mechanism for re-entry into debug mode to continue the single step is not clear

By the definition of the single-stepping, the debug mode is re-entered after completion of single-stepping. And it is consider as completed (just like the condition for icount trigger) when :

1. An instruction retires after having been fetched in a privilege mode where the debug is allowed
2. A trap is taken from a privilege mode where the debug is allowed 

It may be worth calling this out in Sdsec.

closed 2025-06-04 by joxie